lab11/nrf5x-base #3: Check if softdevice is programmed

bradjc, 2015-09-30:
Not sure if there is an easy way to do this (or any way) but it would be AWESOME if the flashing scripts would run `make from-scratch` automatically for you if the softdevice hasn't been programmed because I ALWAYS forget to.

adkinsjd, 2015-09-30:
Definitely possible, but it would add the time overhead of doing a JLink
Read of the entire softdevice to every flash and comparing it to the
current softdevice. Unless we write a softdevice checksum somewhere...

brghena, 2015-09-30:
Done. Commit 121778e554289055c8e03b0e1fe3da422a647c48 updates nrf51-pure-gcc-setup, which has been updated to check and upload the softdevice whenever necessary in commit d8dc8bb43f3834b27ca724ebb7c3fb78b674b9fa

closed 2015-09-30 by brghena