lab11/nrf5x-base #6: Add prettier make output

bradjc, 2015-11-24:
It's a little annoying to scroll through all the lines of output when we include so many folders in compilation. Also need to add a way to get the full output back.

ppannuto, 2015-11-24:
In general, the correct way to do this is to suppress output by prepending
commands with '@', e.g.

all:
   @gcc foo.c

will print nothing. Make will automatically respect the variables, VERBOSE
or V, e.g.

make V=1 all

will print all of the output.

bradjc, 2015-12-02:
Added, although I couldn't get `V=1` to show lines hidden by `@`.

closed 2015-12-02 by bradjc

ppannuto, 2015-12-02:
Huh. Apparently V=1 isn't stock make, just something most make tools add /
use. TIL.

The only thing built into make is suppression using `@` and if you want to
see what will be done `make -n` (aka `--dry-run`), which will print what it
will do (ignoring `@`'s), but won't actually execute anything.

Alternatively, `make -s` will ask make to be `silent`.