lab11/nrf5x-base #19: ST-Link support

wldhx, 2016-12-20:
Would you be interested in a PR with some preliminary ST-Link support? I currently have erasing and flashing (and softdevice flashing) functional (that is, no MAC changing or debugging yet).

brghena, 2016-12-20:
Yeah, we would be interested in that. We primarily use the JLink-edu modules here, but supporting other programmers would be great.

We'll have to determine a good method for specifying which programmer to use. Maybe just a define in app makefiles that defaults to JLink. We could have a "Makefile.jlink" and a "Makefile.stlink" and select between them.

wldhx, 2016-12-20:
I'd thought that we would initially just add targets like `flash.stlink` to existing makefiles, and then perhaps check if users' Makefiles have something like `PROGRAMMER = stlink` in them to decide on what happens if just `flash` is called.

Given that a `.jlink` and `.stlink` makefiles are bound to have a lot in common, I'm not sure if separating them too much will work well.

wldhx, 2017-01-27:
For reference, I'm still on this; it's just that I had a loooong break from all things embedded :)

I think I'll change #20 a bit to include "basic" programming, debugging and README and leave "advanced" programming for some latter time as I haven't had to flash the bootloader by hand even once yet.