lab11/nrf5x-base #21: Question: downloaded_test.bin, how it is working

fletort, 2017-01-21:
Someone can explain a little more how this "check" part is working  ?

we are downloading from the target the downloaded_test.bin ?
We are comparing it with softdevice_test.bin ? (where  this last file comes from ? )

then in my case:
 - erase all with erase-all.jlink
 - flash soft_device with flash-softdevice.jlink (flash the binary s110_softdevice.bin)
 - then flash the application (nRF51_app_s110.hex ) with flash.jlink

brghena, 2017-01-21:
The goal of that code is to check if the desired softdevice version is already on the nRF or not. If it is not, we upload the softdevice.

This works by first downloading 16 bytes at address `SOFTDEVICE_TEST_ADDR` from the flash of the nRF into the file downloaded_test.bin. Next, the file softdevice_test.bin is created by using dd to copy 16 bytes from the same address of the softdevice binary. Finally the two binary files are compared with diff to see if they are identical or not. If they differ, the Makefile calls `$(MAKE) flash-softdevice` which runs erase-all, then flash-softdevice. Finally after that is all complete, the application is uploaded.

So, the steps you are experiencing are correct. However, the second time you flash the nRF, it should detect that the desired softdevice is already on the nRF and only flash the application.

Note: `SOFTDEVICE_TEST_ADDR` was selected through experimentation. At the time the code was written, the 16 bytes at that address differed in every softdevice version. It's possible that it will need to be changed for future softdevice versions.

fletort, 2017-01-22:
thanks for the explication !

ppannuto, 2017-01-22:
@brghena would it be good to move some of this explanation into a comment in that Makefile?

fletort, 2017-01-23:
Sorry, i close the issue, but i was thinking at the same thing :-)

brghena, 2017-01-23:
Good call. Added to make/README.md in d12cb52600d4cca4963283292678b41d3383599a

closed 2017-01-23 by brghena