lab11/nrf5x-base #24: simple_ble does not support SDK 12

wldhx, 2017-01-27:
I'm either missing something or `app_trace` on which `simple_ble` depends has apparently been removed from SDK 12.
```
  CC        ../../make/../lib/simple_ble.c
../../make/../lib/simple_ble.c:19:23: fatal error: app_trace.h: No such file or directory
 #include "app_trace.h"
                       ^
```

brghena, 2017-01-27:
Good question. I should be able to look into this tomorrow or the next day.

fletort, 2017-01-28:
@wldhx maybe look to last commit of my fork (https://github.com/Tech4Race/nrf5x-base/commit/1e9f29d84f92fb1d30ae2764b4833ba0b837a35a) where i success to compile a simple example for nrf52 and SDK12. I didn't Pull request it yet, as i don't know if it is well finish.
A lot of the modification are for the nrf52, but i also add this include to compile nordic example with SDK12:
`LIBRARY_PATHS += $(wildcard $(SDK_INCLUDE_PATH)boards/)`

JakeInvents, 2017-02-05:
@fletort Could you provide a bit more guidance on how to get things compiling for SDK12? I am using your latest commit and am getting the following error when trying to make the blink app.

`main.c:8:21: fatal error: nrf_sdm.h: No such file or directory
 #include "nrf_sdm.h"
`

wldhx, 2017-02-05:
@fletort @JakeInvents +1, actually - I hadn't replied here since I didn't have the time to look into it further than just stopping at the compile error, but if it's not just me... :)

fletort, 2017-02-06:
On my side, i success to compile simple example from Nordic.
I didn't try example from the repo. I will have a look to it.

fletort, 2017-02-06:
Ok, which softdevice are you using ?
In sdk12.2 only s130 & s132 contains the header file `nrf_sdm.h`

[image]

wldhx, 2017-02-06:
@fletort We were compiling the `apps/blink` example which doesn't explicitly specify the softdevice: it seems that your patches on configuration management / guessing broke it.

Upon specifying `SOFTDEVICE_MODEL = s130`, though, I get
```
/usr/lib/gcc/arm-none-eabi/6.3.0/../../../../arm-none-eabi/bin/ld: cannot open linker script file ../../make/../make/ld/gcc_nrf51_s130_WAITED_SOFTDEVICE_VERSION_16_256.ld: No such file or directory
```
So seems like `WAITED_SOFTDEVICE_VERSION` isn't getting substituted. Same with `s130-beacon`, BTW.

fletort, 2017-02-06:
Yes, i make a patch as it was impossible to not use softdevice when compiling with latest sdk.
But your trace show me a bug inside that patch :-)

fletort, 2017-02-06:
I made a fast correction of the bug on my merge request #25.
My last commit on my fork is also updated.
I will try to compile your example to see how the problem is now.

fletort, 2017-02-07:
I success to compile your example (blink) with my correction on my commit.
But you need to add `SOFTDEVICE_MODEL = s130` inside the makefile. 
Prior to my patch, this last one was forced by the global makefile, which is not the normal behaviour.

fletort, 2017-02-07:
@wldhx : which example i can try to compile to see problem on the potential problem on the `simple_ble` lib ?

wldhx, 2017-02-07:
@fletort I don't think there is a functional example out there already, but modifying `s130-beacon` as following should do it:
```
diff --git a/apps/s130-beacon/Makefile b/apps/s130-beacon/Makefile
index 0a836dc..ff165af 100644
--- a/apps/s130-beacon/Makefile
+++ b/apps/s130-beacon/Makefile
@@ -16,7 +16,7 @@ APPLICATION_SRCS += simple_ble.c
 APPLICATION_SRCS += simple_adv.c
 
 SOFTDEVICE_MODEL = s130
-SDK_VERSION = 11
+SDK_VERSION = 12
 
 LIBRARY_PATHS += . ../../include
 SOURCE_PATHS += ../../src
diff --git a/apps/s130-beacon/sdk_config.h b/apps/s130-beacon/sdk_config.h
new file mode 100644
index 0000000..9316362
--- /dev/null
+++ b/apps/s130-beacon/sdk_config.h
@@ -0,0 +1,3724 @@
+
+
+#ifndef SDK_CONFIG_H
+#define SDK_CONFIG_H
+// <<< Use Configuration Wizard in Context Menu >>>\n
+#ifdef USE_APP_CONFIG
+#include "app_config.h"
+#endif
+// <h> nRF_BLE
+
+//==========================================================
+// <q> BLE_ADVERTISING_ENABLED  - ble_advertising - Advertising module
+
+
+#ifndef BLE_ADVERTISING_ENABLED
+#define BLE_ADVERTISING_ENABLED 1
+#endif
+
+// <q> BLE_DTM_ENABLED  - ble_dtm - Module for testing RF/PHY using DTM commands
+
+
+#ifndef BLE_DTM_ENABLED
+#define BLE_DTM_ENABLED 0
+#endif
+
+// <q> BLE_RACP_ENABLED  - ble_racp - Record Access Control Point library
+
+
+#ifndef BLE_RACP_ENABLED
+#define BLE_RACP_ENABLED 0
+#endif
+
+// <q> NRF_BLE_QWR_ENABLED  - nrf_ble_qwr - Queued writes support module (prepare/execute write)
+
+
+#ifndef NRF_BLE_QWR_ENABLED
+#define NRF_BLE_QWR_ENABLED 0
+#endif
+
+// <q> PEER_MANAGER_ENABLED  - peer_manager - Peer Manager
+
+
+#ifndef PEER_MANAGER_ENABLED
+#define PEER_MANAGER_ENABLED 0
+#endif
+
+// </h>
+//==========================================================
+
[... 3651 lines cut ...]
+// <o> SEGGER_RTT_CONFIG_MAX_NUM_UP_BUFFERS - Size of upstream buffer.
+#ifndef SEGGER_RTT_CONFIG_MAX_NUM_UP_BUFFERS
+#define SEGGER_RTT_CONFIG_MAX_NUM_UP_BUFFERS 2
+#endif
+
+// <o> SEGGER_RTT_CONFIG_BUFFER_SIZE_DOWN - Size of upstream buffer.
+#ifndef SEGGER_RTT_CONFIG_BUFFER_SIZE_DOWN
+#define SEGGER_RTT_CONFIG_BUFFER_SIZE_DOWN 16
+#endif
+
+// <o> SEGGER_RTT_CONFIG_MAX_NUM_DOWN_BUFFERS - Size of upstream buffer.
+#ifndef SEGGER_RTT_CONFIG_MAX_NUM_DOWN_BUFFERS
+#define SEGGER_RTT_CONFIG_MAX_NUM_DOWN_BUFFERS 2
+#endif
+
+// </h>
+//==========================================================
+
+// </h>
+//==========================================================
+
+// <<< end of configuration section >>>
+#endif //SDK_CONFIG_H
+
```

brghena, 2017-02-08:
@wldhx I managed to get `s130-beacon` working on SDK12 and pushed changes to master (32d95d3a9cc17624071dfa21206cdfb40b54827f)

There were a couple of small changes that needed to be made to make it compile. Then I found that the RAM size was set incorrectly for it to run. Works on my test hardware now. This hopefully solves your issue, but if not let me know and we can figure out what we're still missing.

wldhx, 2017-02-08:
@brghena Alright, thanks - hopefully my batch of all kinds of Nordics [0] will have arrived by tomorrow - I'll have some testing to do :)

[0]: namely `nrf51822`, `nrf52832` and even two `nrf52840` devkits.

brghena, 2017-02-08:
Good luck! We've only used the nrf51822 ourselves, but are happy to accept PRs to support the 52832 and 52840 if there are problems

JakeInvents, 2017-02-08:
@brghena I can confirm `s130-beacon` is working. Thanks!

I am still having trouble with `service-example ` though. Any ideas? It complains about the sdk_config.h so I moved the file from `s130-beacon` and added `.` to the the library path. Now it's still complaining about `bootloader_types.h`

JakeInvents, 2017-02-08:
@brghena Managed to get a little further by removing the `ENABLE_WIRELESS_DFU` flag in the Makefile.

brghena, 2017-02-08:
Yup, looks like something about DFU changed in SDK12.

You'll also need to add additional APPLICATION_SRCS to the Makefile. For starters these:

```
APPLICATION_SRCS += app_error.c
APPLICATION_SRCS += app_util_platform.c
APPLICATION_SRCS += nrf_drv_clock.c
APPLICATION_SRCS += nrf_drv_common.c
APPLICATION_SRCS += ble_dis.c
APPLICATION_SRCS += ble_srv_common.c
```

I found these by grep-ing the `sdk/nrf5_sdk_12.2.0/` for each undefined function.

You'll also have to enable the BLE device information service module (https://github.com/lab11/nrf5x-base/blob/master/sdk/nrf5_sdk_12.2.0/components/ble/ble_services/ble_dis/ble_dis.c#L18). I'm not sure how yet, possibly in `sdk_config.h`?

I'm out of time for right now, but can take a deeper look in a couple hours.

brghena, 2017-02-08:
Looks like you do NOT need to add `APPLICATION_SRCS += ble_dis.c` (it's automatically added by the Makefiles if you add a `FW_REVISION` or `HW_REVISION` to your Makefile.

Instead, just edit the `sdk_config.h` to set `BLE_DIS_ENABLED` to 1.

Final diff:
```
+APPLICATION_SRCS += app_error.c
+APPLICATION_SRCS += app_util_platform.c
+APPLICATION_SRCS += nrf_drv_clock.c
+APPLICATION_SRCS += nrf_drv_common.c
+APPLICATION_SRCS += ble_srv_common.c

 APPLICATION_SRCS += simple_ble.c
 APPLICATION_SRCS += simple_adv.c
 APPLICATION_SRCS += led.c

-LIBRARY_PATHS += ../../include
+LIBRARY_PATHS += . ../../include
 SOURCE_PATHS += ../../src

-SOFTDEVICE_MODEL = s110
+SDK_VERSION = 12
+SOFTDEVICE_MODEL = s130

 RAM_KB = 32
 FLASH_KB = 256
@@ -22,7 +28,5 @@ FLASH_KB = 256
 HW_REVISION = 2.0.1
 FW_REVISION = 3.4.5

-ENABLE_WIRELESS_DFU = 1
```

(add add an `sdk_config.h`, of course)

wldhx, 2017-02-16:
@brghena I managed to get my code to compile with the exception of UART, in which I didn't look too much yet. I think we should close this issue for now (simple_ble *does* work now :smile:), and I'll just open additional ones if necessary.

Unfortunately, due to *logistics* I still haven't gotten my hands on the newer Nordics; I'll surely play around with them when I can though.

Cheers, and thanks for your work.

closed 2017-02-16 by wldhx