lab11/nrf5x-base #38: (nrf52832 branch) trying to build uart app, multiple definition of UARTE0_UART0_IRQHandler?

akohlsmith, 2018-08-11:
Using commit:4c13fd06 (before the Makefile rewrite that seems to be currently broken).

I'm trying to port an SDK12 application on nrf51 over to nrf52.

sdk_config.h seems to need both UARTE and UART enabled or it just flat out won't build.

```
NRFX_UARTE_ENABLED 1
NRFX_UARTE0_ENABLED 1
NRFX_UART_ENABLED 1
NRFX_UART0_ENABLED 1
```

This seems wrong, but ok.  Everything builds now, but fails to link, claiming that there are multiple definitions of UARTE0_UART0_IRQHandler (in both nrfx_uarte.c and nrfx_uart.c).

I'm getting lost in the C preprocessor macro maze trying to resolve this.

I've added the following in my Makefile:

```
APPLICATION_SRCS += app_fifo.c
APPLICATION_SRCS += app_uart_fifo.c
APPLICATION_SRCS += nrf_drv_uart.c
APPLICATION_SRCS += nrfx_uart.c
APPLICATION_SRCS += nrfx_uarte.c
```

If I leave any of those out, it fails to build due to missing functions.

Any guidance would be very much appreciated.

brghena, 2018-08-14:
If you're still interested, as of 6e52b35 I consider the nrf52832 branch to be about ready for basic use. Particularly, there is an example of how to build an application with board-specific configuration files: https://github.com/lab11/nrf5x-base/blob/nrf52832/apps/demo_nrf52dk/main.c

This should help with the problem of having to constantly add additional C sources to your Makefile. Now you can add them to the board-specific Makefile, and hopefully will need to less and less for various applications. If we find that all boards end up using the same C sources, I might end up pulling them all the way into the build system.

Notably missing from the branch as it currently is, is wireless example applications of any kind. I'm betting that there are several BLE sources, libraries, and configurations missing at the moment. That is of a little lower priority for me right this moment since it's not on the critical path for the class I'm working on, but will come in the next few weeks.

akohlsmith, 2018-08-21:
Absolutely still interested! Just getting back to this, I will check it out shortly. Thank you so much for all of your hard work on this build system. It's by far the best way to develop on nRF that I've run across.

brghena, 2018-08-21:
Awesome. Let me know if you have any problems. Documentation is still tragically missing, but there are some good examples. For another good example, we've also been using it out-of-repo here: https://github.com/lab11/buckler

akohlsmith, 2018-08-28:
This is great, thank you so much.

The only thing I'm currently fighting is how to properly disable the softdevice. The config defaults to s132 but if I leave it blank (by changing the default in `make/Configuration.mk`) then I get errors because it's not finding nrf_error.h. I haven't spent a lot of time working though this yet but I suspect this is an edge case that your work hasn't focused on.

brghena, 2018-08-28:
Yeah, the confusion is totally a fault of the lack of documentation so far. Sorry about that.

To use no softdevice add "SOFTDEVICE_MODEL = blank" to the app Makefiles. I made the decision to try to be explicit about which softdevice (or lack thereof) you're using instead of just leaving the variable undefined.

ppannuto, 2018-08-28:
You should be able to add something like

ifeq ($(SOFTDEVICE_MODEL),)
info You must specify a soft device
info valid options are...
error No softdevice specified
endif

To catch/enforce this

akohlsmith, 2018-09-20:
Do you have any suggestions or examples where I can use both UARTs on 52840? I'm trying to use one of them for regular old retargeting (printf() etc.) and the other for talking to another device. I see some examples on Nordic's site and one in particular that uses both (https://devzone.nordicsemi.com/f/nordic-q-a/32120/how-to-use-gpio-port-1-for-uarts) but it doesn't want to build under nrf5x.

I *think* it's tied to this UART/UARTE confusion again.

```#define UART_ENABLED 1
#define UART_EASY_DMA_SUPPORT 1
#define UART_LEGACY_SUPPORT 1```

will error out because nrf_drv_uart.h can't find the UARTE typedefs (e.g. nrf_uarte_baudrate_t, nrf_uarte_hwfc_t, etc.).

If I instead try to force UARTE (I still am not sure the difference between UART and UARTE, but nrf528240_peripherals.h says it has two UARTEs and only one UART), I get the same issue. That sdk_config.h blurb is as follows:

```#define UART_ENABLED 1
#define NRFX_UARTE_ENABLED 1
#define NRFX_UART_ENABLED 0```

(I tried forcefully disabling NRFX_UART because the error is coming from line 116 of nrf_drv_uart.h and that is only invoked if NRF_DRV_UART_WITH_UART is not defined. The top of that file creates either NRF_DRV_UART_WITH_UARTE or _WITH_UART.

akohlsmith, 2018-09-20:
I'm going to close this issue and open a new one; I think I've figured out the issue and it's that on the nRF52840 there are two UARTEs but the `nrf52840_peripherals.h` file says there are two UARTEs and a single UART. This isn't entirely correct -- the UART is at address 0x40002000, and UARTE0 is at the same address (UARTE1 is at 0x40028000).

BTW UARTE is just a UART that supports EasyDMA. That's it.

Anyway -- since `nrf52840_peripherals.h` says there is a UART and two UARTEs, the macros at the top of `nrf_drv_uart.h` will define both `NRF_DRV_UART_WITH_UARTE` and `NRF_DRV_UART_WITH_UART`, which causes the build to fail because there's no way to say "two UARTEs and one UART" so it tries to instantiate two UARTs.

closed 2018-09-20 by akohlsmith