lab11/nrf5x-base #39: How to build for both SDK12 (nrf51) and SDK15 (nrf52)?

akohlsmith, 2018-08-28:
Less an issue, but not sure how else to raise the question.

I'd like to be able to develop for both nrf51 and nrf52 devices without having two copies of nrf5x-base checked out. I'm not trying to use the same source code to build for both (although that would be nice) but rather just to have `apps/nrf51_foo` and `apps/nrf52_bar` and be able to build both using the nrf5x-base submodule.

Is this an eventual goal? I understand that nrf52 support is currently in its own branch. Is the intention to eventually have both supported in master?

brghena, 2018-08-28:
Good question. The real answer is that we haven't decided yet.

So far, my thoughts had been on the keeping them separate side of the world. While supporting both SDK12 for the 51 and SDK15+ for the 52 would be good, I'm worried that it will create a lot of difficulties in getting everything working together because of the differences in the SDKs. It's fairly frustrating that the newer SDKs no longer support the nRF51s even though they're still in wide use.

One option is that we can have a tagged version of this repo that supports nRF51s, and master that supports nRF52s. To have one project use both, you'd create two different submodules that point to different commits.

A different possibility is that when we're happy with the nRF52832 branch, we can attempt to pull support for SDK12 into it, and maybe it won't be that hard?

akohlsmith, 2018-08-30:
I'd definitely like to have one happy submodule, but it looks like a not-insignificant amount of work to bring the sdk12 stuff up to the new build environment (with boards/, etc.).

In the meantime I'm trying to detect which SDK I'm building under so I can use as much common source as possible. I thought it'd be a simple task of

```
#if (SDK_VERSION >= 15)
/* SDK15+ stuff here */
#else
/* SDK12 stuff here */
#endif

```
but while that works for SDK12, it doesn't appear to work for SDK15 (which executes the else branch). Fine, I thought I'd just look at NRF_MODEL and do it that way, but that works for SDK15 but not 12.

Oddly, make -pn seems to show the correct output for SDK_VERSION and NRF_MODEL for both SDK12 and SDK15, but it's not getting picked up in the code within libraries.

acalatrava, 2021-05-12:
I guess this can be closed too #44

closed 2021-05-12 by brghena