lab11/nrf5x-base #40: (nrf52832 branch) - cannot use two UARTEs

akohlsmith, 2018-09-20:
On the nRF52840 there are two UARTEs but the `nrf52840_peripherals.h` file says there are two UARTEs and a single UART. This isn't entirely correct -- the UART is at address 0x40002000, and UARTE0 is at the same address (UARTE1 is at 0x40028000).

Anyway -- since `nrf52840_peripherals.h` says there is a UART and two UARTEs, the macros at the top of `nrf_drv_uart.h` will define both `NRF_DRV_UART_WITH_UARTE` and `NRF_DRV_UART_WITH_UART`, which causes the build to fail because there's no way to say "two UARTEs and one UART" so it tries to instantiate two UARTs.

How to recreate:
Set up a project where you want to use two UARTEs. I believe the correct `app_config.h` stanzas are as follows:

```
#define UART_ENABLED 1
#define UART0_ENABLED 1
#define UART1_ENABLED 1
#define UART_EASY_DMA_SUPPORT 1
```

In the SDK, `apply_old_config.h` will take these defines and define `NRFX_UARTE_ENABLED`, `NRFX_UARTE0_ENABLED` and `NRFX_UARTE1_ENABLED`. `nrf_drv_uart.h` will set up for two UARTEs since `UARTE_PRESENT` is defined in nrf52840_peripherals.h` and also because `apply_old_config.h` created `NRFX_UARTE_ENABLED`

This is all perfect until you realize that `nrf_drv_uart.h` also defines `NRF_DRV_UART_WITH_UART` since the peripherals.h file says there is a UART as well as two UARTEs. 

I believe this is an error, as the nRF52840 has a total of two UART devices with DMA, not three (two with DMA and one without). I think the correct fix is to modify `nrf52840_peripherals.h` but before I create a PR I wanted to run this by you.

brghena, 2018-09-21:
Wow, that sounds like a big pain to think about. Good job debugging that. I agree that there are two UART/UARTE peripherals (not three). However, this bug is in the Nordic SDKs, not in any of our files. As a rule we try not to make any edits directly to the Nordic SDK files.

You should try reporting this problem to [Nordic devzone](https://devzone.nordicsemi.com/) and see what Nordic has to say about it. (If you do so, post a link here too)

mhazley, 2019-01-22:
Hi @akohlsmith did you ever get a resolution to this? I am using SDK 15.0.0 on 52840 and having trouble configuring both UART!