lab11/nrf5x-base #43: device not visible when using ENABLE_WIRELESS_DFU=1

neu-rah, 2019-08-05:
even examples that already have ENABLE_WIRELESS_DFU=1, as `service-example` render device not visible (not advertising?).
Removing that line everything works ok.
I'm using nRF51822xxaa (16K, 256K)

just changed:
```makefile
NRF_IC=nrf51822
RAM_KB=16
FLASH_KB=256
```

nealjack, 2019-08-07:
Hi, that is a weird issue. It's been a long time (and many SDKs) since I've touched the BLE DFU stuff, but if my memory serves right, you are either not enabling sufficient peripheral/central roles in the SDK config, or allocating enough memory for them in your linker script. This was always a very confusing process that I'm afraid we didn't document properly.

Unfortunately, like the nRF SDK support for nRF51x, this repository is not in active development. We've moved to the newer nRF52x platforms (and a new [repo!](https://github.com/lab11/nrf52x-base)). If you want to dig into the underlying issue here and submit a pull request, we'd be glad to look it over and pull it in.

neu-rah, 2019-08-07:
hi!
thanks, probably one of those things. I followed the instructions on the README.md and just using `service-example` as my app, copied to /app and edited Makefile:

added:
 ```Makefile
RAM_KB=16
FLASH_KB=256
```

changed:
```Makefile
NRF_BASE_PATH ?= ../../nrf5x-base
```

files are found, things look ok but `make` still lists 32K chip version
```text
make
BUILD OPTIONS:
  SoftDevice  s110
  SDK         10
  nRF         nrf51822
  RAM         32 kB
  FLASH       256 kB
  Board       BOARD_NULL
 
```

so maybe `apps/service-example/Makefile` is not the place for adding this?

putting this ram and flash size on `nrf5x-base/make/Makefile` seems to solve it, build now all say 16k/256k, so i must be missing something

nealjack, 2019-08-07:
Make sure you aren't redefining `RAM_KB` twice.

I just tried to build it and saw something similar. So if the `ENABLE_WIRELESS_DFU` option is included in your makefile (defined at all), you are also building the bootloader, and you should see two `BUILD OPTIONS` lists, one that lists the correct settings for the `service-example` application, and another for the bootloader, which seems to always list the 32kB.

I'd recommend not using the bootloader or DFU feature. That should solve your problem.

neu-rah, 2019-08-07:
indeed, but avoiding BLE DFU is not an option, I'm only here because of it...
thanks for the advice. I will let you guys know if i find something worth of note, and please if someone has some relevant info about this please let me know.

closed 2019-08-07 by neu-rah

nealjack, 2019-08-07:
I think you'll just need to edit the makefiles and linker scripts to build the DFU using the smaller memory option. Back when I was working on this feature, I think I ran into troubles using the 16kB option as there was not enough memory.