lab11/nrf5x-base #44: Compile for nRF51 and nRF52

acalatrava, 2021-04-21:
Does anyone knows if it’s possible to compile using SDK11 to both platforms?

brghena, 2021-04-21:
SDK11 _should_ support the nRF51822 and nRF52832. It's pretty out of date though.

Since the capabilities of the nRF51 and nRF52 are different, we ended up creating a whole new repo for nRF52 stuff, which is here: https://github.com/lab11/nrf52x-base It's still in active use (unlike this repo).

acalatrava, 2021-04-21:
Yeah I know there is the nrf52x-base, which is great, however I expected to create a single program and be able to compile it on both platforms. Since SDK11 is the only one targeting both platforms and this project’s name is nrf5x-base I expected to be able to do it... :-/

By setting the softdevice to 132 and NRF to nRF52 I get an error telling that there is no tule able to compile system_nrf52.o 

Any hint is welcomed!

brghena, 2021-04-21:
Oh, I understand. Unfortunately, this repo was made before the nRF52 series came out... which makes the name inappropriate now.

As for the error. I'm not very surprised that build system here isn't stable across a lot of different configurations. Generally the approach is that you'll have to edit `make/Makefile` to include proper paths given the configurations you've made (usually based on the SDK version). To find the files it's looking for, you can `find` and `grep` for them in `sdk/nrf51_sdk_11.0.0/`. You'll have to play around with it and be willing to dig to make things work.

If you find somethings that work and want to make a PR, we'll probably merge it into master. But more or less this is a dead repo since we've moved on from the nRF51822. You're welcome to fork it and make your own changes as necessary too.

acalatrava, 2021-05-06:
Well, it wasn’t that hard 👍 #45

brghena, 2021-05-09:
Closed via #45

closed 2021-05-09 by brghena