lab11/nrf5x-base #48: ADC reading

beckmx, 2023-09-12:
Hello there, I know it's a little bit old, but was wondering if there is a tiny example on how to read adc from a pin, I tried adding these headers but failed to compile.

```
#include "nrf_log.h"
#include "nrf_log_ctrl.h"
#include "nrf_log_default_backends.h"
```

Greetings!

brghena, 2023-09-15:
Hard to say what issue you're running into without seeing the error message. There are a couple of examples you could look at though.

There is an example app for ADC, although it's pretty rough code. https://github.com/lab11/nrf5x-base/blob/master/apps/adc-test/main.c

There are a couple of better examples external to this repo, that rely on this as a submodule. This is probably the most complete one: https://github.com/lab11/buckler/blob/master/software/apps/analog_accelerometer/main.c

I have a repo I use for teaching that has some partial examples. This breadboarding example is mostly there, although it has some "TODO" comments where I expect students to implement stuff on their own: https://github.com/nu-ce346/nu-microbit-base/blob/main/software/apps/breadboard/main.c There's also this example which reads an entire buffer of samples from a microphone: https://github.com/nu-ce346/nu-microbit-base/blob/main/software/apps/record_and_play/main.c (the analog code is complete).

closed 2023-09-15 by brghena

beckmx, 2023-09-21:
Hello there, before you commented I was reading through the repo to glue some examples together, I managed to compile everything but the adc doesnt answer me :(, if you have some suggestions or spot something, here is the code i used:

```

//original
#include <stdbool.h>
#include <stdint.h>
#include <stdio.h>
#include "app_uart.h"
#include "app_error.h"
#include "nrf_delay.h"
#include "nrf_drv_config.h"

#include "led.h"
// addons
#include "nordic_common.h"
#include "softdevice_handler.h"
#include "nrf_drv_adc.h"
#include "app_util_platform.h"

// 
#define LED0 7

#define MAX_TEST_DATA_BYTES  (15U)
#define UART_TX_BUF_SIZE     256
#define UART_RX_BUF_SIZE     256

#define ADC_BUFFER_SIZE 10                                /**< Size of buffer for ADC samples.  */
static nrf_adc_value_t       adc_buffer[ADC_BUFFER_SIZE]; /**< ADC buffer. */
static nrf_drv_adc_channel_t m_channel_config = NRF_DRV_ADC_DEFAULT_CHANNEL(NRF_ADC_CONFIG_INPUT_4); /**< Channel instance. Default configuration used. */

void uart_error_handle (app_uart_evt_t * p_event) {
    if (p_event->evt_type == APP_UART_COMMUNICATION_ERROR) {
        APP_ERROR_HANDLER(p_event->data.error_communication);
    } else if (p_event->evt_type == APP_UART_FIFO_ERROR) {
        APP_ERROR_HANDLER(p_event->data.error_code);
    }
}

/**
 * @brief ADC interrupt handler.
 */
static void adc_event_handler(nrf_drv_adc_evt_t const * p_event)
{
    if (p_event->type == NRF_DRV_ADC_EVT_DONE)
    {
        uint32_t i;
        for (i = 0; i < p_event->data.done.size; i++)
        {
            printf("adc: %d\n",p_event->data.done.p_buffer[i]);
            if(p_event->data.done.p_buffer[i] > 512)
                led_on(LED0);
            else
                led_off(LED0);
            //NRF_LOG_PRINTF("Current sample value: %d\r\n", p_event->data.done.p_buffer[i]);
        }
    }
}
/**
 * @brief ADC initialization.
 */
static void adc_config(void)
{
    ret_code_t ret_code;
    nrf_drv_adc_config_t config = NRF_DRV_ADC_DEFAULT_CONFIG;

    ret_code = nrf_drv_adc_init(&config, adc_event_handler);
    APP_ERROR_CHECK(ret_code);

    nrf_drv_adc_channel_enable(&m_channel_config);
}

int main (void) {
  led_init(LED0);
  led_on(LED0);
  adc_config();

    uint32_t err_code;
    const app_uart_comm_params_t comm_params = {
          RX_PIN_NUMBER,
          TX_PIN_NUMBER,
          0,
          0,
          APP_UART_FLOW_CONTROL_DISABLED,
          false,
          UART_BAUDRATE_BAUDRATE_Baud115200
      };

    APP_UART_FIFO_INIT(&comm_params,
                         UART_RX_BUF_SIZE,
                         UART_TX_BUF_SIZE,
                         uart_error_handle,
                         APP_IRQ_PRIORITY_LOW,
                         err_code);
    APP_ERROR_CHECK(err_code);

    printf("Start: \n");

    while (true) {
        uint8_t cr;
        while (app_uart_get(&cr) != NRF_SUCCESS);
        printf("adc: \n");
        // manually trigger ADC conversion
        nrf_drv_adc_sample();
        printf("Got: %c\n", cr);
    }
}
```

From what i understand everytime i send a character to the UART console it should read the adc but i think it doesnt read it.

Greetings!